visose/Robots #3: DO commands not executing on UR10

AndreaRossi1988, 2017-01-04:
When adding DO commands on targets, they do not seem to execute when the program is uploaded and launched via the RemoteUR component. The DO commands execute correctly when placed at the beginning of the program in the Program component, but not when placed on the Target component.

visose, 2017-01-04:
Hi Adrea, thank you for inaugurating the issues section of this repository.
Make sure you're using the latest version of the plugin. I used the DO commands with the RemoteUR a few weeks ago and it was working correctly. If it's still not working please post a simplified Grasshopper definition with the issue and I'll have a look.

ar0551, 2017-01-05:
Hi Vicente, thanks for the quick reply!
**First problem solved**: I was using the release 2, with the newer one DO commands work.
Second problem: I still have problems with the DO commands. If I connect to the robot, upload the code and then disconnect immediately, the execution works fine and the DO commands are executed correctly. However, if I mantain the connection to the robot open (eg. to use the play-pause options) and re-upload the same code, the DO commands do not always work, but sometimes get skipped, or exectuted just when I disconnect from the robot. Does this make sense? I can send you a file or explain better if I haven't been clear.
Thanks a lot.

visose, 2017-01-05:
You explained it clearly but this has never happened to me. When keeping the connection open I've experienced some other bugs, like parts of the pendant's UI not responding properly but never the DO commands of the streamed program not executing. Tomorrow I'll be testing some stuff in a UR so I can have a look. If you can, upload a simplified version of the problem so I can test it.

ar0551, 2017-01-05:
Thank you, here is the file where I had the problem: https://app.box.com/s/kwa99o72dpw46t266idw0s2qdf1z514y

The problem happens has follows:
- I connect to the robot and upload the code once, and the execution is correct with all the DO commands
- I upload the code a second time, and this time sometimes (really strange!) the DO commands do not work. I also noticed that, when they are skipped, they get executed once I disconnect from the robot.

Reading what you wrote, I start thinking that the DO signals might get triggered, but their status might not get updated on the pendant UI. I had nothing connected to the DO signals, so the only check I had was the UI. I will try next week with a signals actually triggering a gripper, and will let you know.
Meanwhile, thank you!

visose, 2017-01-06:
It's likely that, as you say, the display is not updating but the IO turns on and off correctly. The pendant receives data through the same network interface that's being used by the plugin (the secondary client interface). I'll try to figure out a way around this issue.

visose, 2017-01-31:
In the next release the RemoteUR component will connect, send a message and then disconnect (you won't have to manually connect first). This will avoid the pendant's UI from not updating data properly.

closed 2017-01-31 by visose

ar0551, 2017-01-31:
Great, thanks a lot!