visose/Robots #4: Can't load Robot System

vsardenberg, 2017-01-31:
Hello!

I'm trying to use ROBOTS with our UR-5 in Leibniz University in Hannover.
When I drag and drop Load Robot and plug a value list on it the component becomes red and I receive this error on the balloon:

1. Solution exception:'>' is an unexpected token. The expected token is '='. Line 45, position 94.

Even if I open simple example.gh, the same thing happens. I've copied all libraries to C:\Users\Victor Sardenberg\Documents\Robots\

Any ideas?

Thanks,

Victor Sardenberg

visose, 2017-01-31:
Hi Victor, it might have to do with the xml you're using for the definition of the UR5. Could you send it to me? (both the xml and the 3dm that you have in the \Robots\ subfolder)

There's a pull request by Ryan Luke Johns that contains a UR5 (https://github.com/visose/Robots/pull/2) but it has to be updated to work on the latest version of the plugin. I can try to add it when I get some free time.

vsardenberg, 2017-01-31:
Hello visose,

Thanks for your reply. I downloaded all .xml files from this github and I didn't change anything. 
Here you have a link to a copy of my folder:

https://www.dropbox.com/sh/hhd15wy8xr5oev5/AAD_0vbcv8_XCUnDsTeUCz27a?dl=0

Thanks

visose, 2017-02-02:
Ah, so not even the default libraries are working? Did you follow the instructions in the wiki to install the plugin? Make sure you follow all the points, including the running the Rhino command "GrasshopperDevelopperSettings" and the one about unblocking the .gha and .dll files.

vsardenberg, 2017-02-02:
Hello Vicente,

I followed every step and double-checked it.

Victor Sardenberg
www.victorsardenberg.com

visose, 2017-02-02:
That error seems to be from a badly formatted xml file. It's possible you didn't download them correctly (you downloaded GitHub's html file rather than the xml itself. Go to the library,  click on the xml file and then click on the "raw" button (or right click on that button, save as).

I uploaded a library for the University of Leuven that contains a UR5.

vsardenberg, 2017-02-02:
Thank you very much. Now everything works nicely.

Victor Sardenberg
www.victorsardenberg.com

closed 2017-02-05 by visose