visose/Robots #7: Tutorial on creating own robot xml + rhino file?

yijiangh, 2018-06-19:
Hi @visose ,

Thank you for creating such a wonderful package! I am wondering if it's possible to document a bit more on creating one's own robotcell (xml + rhino file)?

From my observation (snapshot below on the Bartlett robotcell), it seems like the xml and the geometry data in the rhino file is linked by layer's name, e.g. the layer in rhino should be `RobotArm.<manufacturer>.<robot model>` and the xml file and rhino file are linked by the same file name, right?

[image]

Apart from this, I am confused in the following three parts:

(1) In the xml file, for each joint we have `<Revolute number="1" a ="25" d ="400" minrange = "-170" maxrange ="170" maxspeed ="360"/>`. What do `a` and `d`'s value mean?

(2) Which parameter in the xml file specify the current robot's joint pose in the rhino file? (like the robot pose in the picture above, it is definitely not the all-zero configuration)

(3) If I want to configure a 6-axis robot on a linear track, how should I do this? Should I create a new robotcell entry in the xml file to specify a 6+1=7 axis robot, or should I just treat the linear track as the second robot system, and input it, together with the 6-axis robot, in an ordered list to the `create program` component?

Finally, if I'd love to use this package in an academic project, what would be the preferrable way to cite this package?

Thank you so much,
Yijiang

visose, 2018-06-19:
Hi yijiangh, if you tell me the robot models and track model you have, and name of your institution, I can prepare a set of .xml and .3dm files for you. This will also help you understand how the xml works in case you need to modify it later. I agree that this should be documented somewhere.

1 - These are the Denavit–Hartenberg parameters (https://en.wikipedia.org/wiki/Denavit–Hartenberg_parameters)
2 - The DH parameters define the relative positions of the joints (as a chain).

3- In the Bartlett .xml file you'll see a how to insert a track. You should have one <RobotCell> per robot controller. If the track and robot are connected to the same controller, you'll have two mechanisms inside one RobotCell, you'll probably also have both of them inside the same Mechanical Group.

You can cite this as: Robots plugin for Grasshopper (https://github.com/visose/Robots)

yijiangh, 2018-06-20:
Hi Vicente,

Thank you for your explanation. I am a grad student at MIT and I am trying to use your package as a simulation tool. The robot is an [IRB 6640-185-280 ](https://search-ext.abb.com/library/Download.aspx?DocumentID=ROB0001EN&LanguageCode=en&DocumentPartId=&Action=Launch) and the track model is the UCLA KUKA model that I get from the KUKA|PRC package. MIT does not currently own these giant machines but I'd love to work with them visually.

The only thing that I am having trouble with is the DH parameters in the xml file - do you have some documentation on how do you generate them? 

I've set up a draft version of the xml and rhino file (attached here: [Robots-MIT_sim.zip](https://github.com/visose/Robots/files/2121127/Robots-MIT_sim.zip)), where most of the data has been filled up except the DH parameters. Could you please help me complete them? For your convenience, I create a sublayer called `joint_points` to mark the joint positions (see snapshot below).

[image]

I am expecting the finished setup to be the same as the following:

[image]

Thank you so much for your patience and help!

visose, 2018-06-21:
Hi, attached is the working modified library. I added some comments in the .xml.
The DH parameters you can get them from robot dimensions (see in comments).
It's a bit weird that you're using a KUKA track with an ABB robot (I would try to find out the actual track model for better naming conventions (rather than naming it UCLA Track) but it will work anyway.
In the Rhino file a robot should have only 7 meshes (base + 6 axes), each in the apropiate layer. I joined the different mesh parts together and applied some mesh decimation (_reducemesh command).

[MIT_sim.zip](https://github.com/visose/Robots/files/2123816/MIT_sim.zip)

yijiangh, 2018-06-21:
Thank you very much! The example definitely helps me better understand how these things should be set up. With this, I will close the issue.

closed 2018-06-21 by yijiangh