visose/Robots #8: Joint configuration error on Bartlett KR6 robot

yijiangh, 2018-06-19:
I am writing to report a joint configuration error when I try to input a single joint configuration target to the Bartlett KR6 robot.  I am using the [fifth release (0.0.7)](https://github.com/visose/Robots/tree/0.0.7)

I find all the joint's lower and upper limit for this KR6 robot in the [bartlett.xml](https://github.com/visose/Robots/blob/master/Libraries/bartlett.xml) file and simply choose the middle value for each joint's lower and upper limit, convert them all from degree to radian, formulate it in text using comma `,` as separator, and plug it into a `joint target` component. However, the `create program` component gives me a really wrong configuration in rhino. A snapshot is shown below to compare the ground truth pose (orange color, generated from KUKA|PRC) and the one generated from the Robots package:

[image]

I am wondering if there is anything that I do wrong here? I suspect that I am not using `joint target` component in a correct way... Any help would be much appreciated!

For your convenience, a grasshopper file is attached here for you to reproduce the error.
[bartletts-robots_joint-error.zip](https://github.com/visose/Robots/files/2113582/bartletts-robots_joint-error.zip)

Thank you!

visose, 2018-06-19:
Hi yijiangh, this is the expected behaviour. The plugin supports 3 different robot manufacturers, each define differently where the position of the angle 0 is and the direction. For the same 6 set of angle values, a KUKA robot will be positioned different than an ABB or UR robot. The plugin intends to abstract manufacturer specific peculiarities. The plugin uses one criteria in radians and this gets converted to each different robot language so that all robots will be positioned the same as in the simulation. Do you need to convert from kuka degrees to the plugins radians?

yijiangh, 2018-06-19:
Hi @visose , thanks for your reply. I'm not sure if I fully understand your reply. Why should radians and degrees be manufacturer dependent? Could you please give me an example explaining this? Let's say, if I want to get the robot to the "middle value" of all the joints, what value should I input into the `joint target` component?

Currently, as in the attached gh file in my post above, I use degrees, use the GH built-in `degrees-to-radians` for conversion, and then construct a text out of these values. Is this incorrect in your plugin's setup?

Thank you!

visose, 2018-06-19:
Think of a circle. What point of the circle is degree 0? It could be the rightmost point, the bottommost point, or any other arbitrary point. What is degree 15? It could be clockwise up from the 0 degree point or counter clockwise down and so on. What I mean is that the angle values you set on the plugin are different than the ones you'll see on the teach pendant (irrespective of the different units of radians vs degrees), but the actual position of the joints will be the same. For the default home position the 6 joints values you should set on the plugin are: 0,0,pi*0.5,0,0,0.

visose, 2018-06-19:
I forgot to mention: If you want to convert from the degrees you see on the teach pendant to the radians you need to use on the plugin, there is a component for that "Degrees to Radians". Make sure you use the component of the Robots plugin, not the default component.

yijiangh, 2018-06-20:
> If you want to convert from the degrees you see on the teach pendant to the radians you need to use on the plugin, there is a component for that "Degrees to Radians". Make sure you use the component of the Robots plugin, not the default component.

Yes, I was using the primitive degrees-to-radians GH component. Switching to the one in the Robots plugin resolves the problem. Thank you so much for your explanation!

closed 2018-06-20 by yijiangh