visose/Robots #14: UR mistake

francescobic, 2019-04-24:
Hi guys

i'm trying to use robot to control an old uR10 cb1.8 but i have a problem with the code, UR controller say me "infinite loop detected". i have solve this problem manually, i have change the definition bringing the code (C) output and replace text >inserti item >and stream content from panel. i have added a line at the end of the code that close the program (line 18).
i can share the definition but is better to solve from "create program" component

here a sample of my code streamed from panel:

0. def Program():
1.   weaverTcp = p[-0.22000, 0.00000, 0.56150, 2.2214, 2.2214, 0.0000]
2.   weaverWeight = 0.200
3.   weaverCog = [-0.22000, 0.00000, 0.56150]
4.   Speed000 = 0.02000
5.   Zone000 = 0.00300
6.   set_tcp(weaverTcp)
  set_payload(weaverWeight, weaverCog)
7.   movej([1.6021, -1.3203, 1.7483, -1.9987, -1.5708, 2.6493], a=3.1416, v= 0.314, r=Zone000)
8.   movel(p[-0.37700, -0.51756, -0.43400, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
9.   movel(p[-0.37700, -0.51756, -0.47800, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
10.   movel(p[-0.37700, -1.26756, -0.43400, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
11.   movel(p[-0.37700, -1.26756, -0.47800, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
12.   movel(p[0.37300, -1.26756, -0.43400, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
13.   movel(p[0.37300, -1.26756, -0.47800, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
14.   movel(p[0.37300, -0.51756, -0.43400, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
15.   movel(p[0.37300, -0.51756, -0.47800, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
16.   movel(p[-0.00200, -0.89256, -0.17100, 0.0000, 0.0000, -1.0472],a=1.00000,v=Speed000,r=Zone000)
17. end
18. Program()

visose, 2019-04-24:
Hi, we have the same UR10 version here at the Bartlett but have never encountered that error.
I added below the minimum code the plugin creates with one target, this should work properly. Are you using the RemoveUR component to stream the program over the network to the controller? This is the recommended way.
```
def Program():
  DefaultToolTcp = p[0.00000, 0.00000, 0.00000, 0.0000, 0.0000, 1.5708]
  DefaultToolWeight = 0.000
  DefaultToolCog = [0.00000, 0.00000, 0.00000]
  DefaultSpeed = 0.10000
  DefaultZone = 0.00000
  set_tcp(DefaultToolTcp)
  set_payload(DefaultToolWeight, DefaultToolCog)
  movej([0.0000, -1.5708, 0.0000, 0.0000, 0.0000, 0.0000], a=3.1416, v= 0.314, r=DefaultZone)
end
```

francescobic, 2019-04-24:
sorry, remoteUR work good, is the file from file saver that have this problem

closed 2019-04-24 by visose