visose/Robots #15: Does it run on Grasshopper with Rhino 6?

boundlessmaking, 2019-04-25:
I am trying to get it to run on rhino 6. But I cant open any of the example files, I always get an error that it couldnt read the file. When I place the "Load robot system" component it also gives an error: 1. Solution exception:'>' is an unexpected token. The expected token is '='. Line 44, position 92.

Im guessing this refers to [line 44 of Util.cs](https://github.com/visose/Robots/blob/e2d7f0037ac721ae55090290186196eb84f0eb70/RobotsGH/Util.cs#L44l)? 
I followed all the other steps of the installation (installing .NET and loading the 3dm and .xml into Documents/Robots) 
Anything else I could check ?
Thanks in advance.

boundlessmaking, 2019-04-25:
Just found this one: https://github.com/visose/Robots/issues/4 will try it first.

boundlessmaking, 2019-04-25:
Okay nevermind I just downloaded the files wrong as described and solved in Issue #4

visose, 2019-04-25:
There's no better feeling that solving a problem by yourself 😋

closed 2019-04-25 by visose

boundlessmaking, 2019-04-25:
Indeed! Very nice plugin by the way you should post it on Food4Rhino.com to make it easier to find. I have been trying all the free plugins there (machina, scorpion, Robots.io) for weeks without much success. putting my hopes into this one now :D