visose/Robots #16: Example with an external acis?

boundlessmaking, 2019-05-05:


visose, 2019-05-05:
[trackExample.zip](https://github.com/visose/Robots/files/3145567/trackExample.zip)
It uses the Bartlett robot library with the KR60 on a track. If you download this library the track might e commented out on the .xml, you'll have to remove the <-- and --> lines.

Attached is a minimal example of moving a robot in a track based on the path.
It moves the robot to keep it aligned with the targets in the direction of the track. You usually would not want to do this as moving the whole robot on every target less desirable that getting close and then moving the rest of the joints to reach a target (since it uses more energy, slower speed/ramping, less precision).

boundlessmaking, 2019-05-05:
Thanks a lot, will try it out.

visose, 2019-05-05:
The previous reply was written before you added the body of your comment. Regarding a positioner, you will use it in a similar way (the external input will be radians rather than millimeters). You can enable kinematic coupling using the 'create frame' component. The frame (and targets attached to it) will move/rotate together with the positioner. If you have a single robot with the positioner as the only external axis, both inputs G and M should be 0.

visose, 2019-05-05:
I understand you want to connect a motor as an external axis to the robot controller so that the controller's computer synchronizes it with the rest of the joints taking it's dynamics into account and so on. I've only done this with motors that are sold by the manufacturer as external axis compatible. This makes the integration a lot easier, but they are expensive. There are ways to synchronize a separate servo that comes with its own controller, but i would avoid this if you can as i think it's adding unnecessary complexity (unless you're doing something like moving a gigantic track that the robot controller wouldn't be able to handle). In any case this will only be possible with closed loop servos, not with stepper motors.

boundlessmaking, 2019-05-05:
My use case is fairly simple: 
I need infinite rotation on the last joint of the robot. To do this I was thinking of adding a DiY version of a rotary table similar to this one to the robot: [image] 

As you have already stated this will add a lot of complexity, especially when it comes to trajectory planning and my knowledge of programming is limited.

boundlessmaking, 2019-05-05:
I am getting errors when trying to use your example: ´Track external axis not configured on this target´ where do I have to configure it ?

visose, 2019-05-05:
Do you see a track when you enable the preview of the 'load robot system' component?

boundlessmaking, 2019-05-05:
> Do you see a track when you enable the preview of the 'load robot system' component?

Yes, it is there, however the robot is shifted relative to the track and not directly on it 
[image]

visose, 2019-05-05:
The whole robotcell should look like this:

    <RobotCell name="Bartlett-KR60 on track" manufacturer="KUKA">
    <Mechanisms group="0">
      <RobotArm model="KR60-HA" manufacturer="KUKA" payload="60">
        <Base x="0.000" y="0.000" z="0.000" q1="-0.707107" q2="0.000" q3="0.000" q4="0.707107"/>
        <Joints>
          <Revolute number="1" a ="350" d ="815" minrange = "-185" maxrange ="185" maxspeed ="128"/>
          <Revolute number="2" a ="850" d ="0" minrange = "-135" maxrange ="35" maxspeed ="120"/>
          <Revolute number="3" a ="145" d ="0" minrange = "-120" maxrange ="158" maxspeed ="128"/>
          <Revolute number="4" a ="0" d ="820" minrange = "-350" maxrange ="350" maxspeed ="260"/>
          <Revolute number="5" a ="0" d ="0" minrange = "-119" maxrange ="119" maxspeed ="245"/>
          <Revolute number="6" a ="0" d ="170" minrange = "-350" maxrange ="350" maxspeed ="322"/>
        </Joints>
      </RobotArm>

      <Track model="KL1000-2" manufacturer="KUKA" payload="1000" movesRobot="true">
        <Base x="0.000" y="0.000" z="0.000" q1="1.000" q2="0.000" q3="0.000" q4="0.000"/>
        <Joints>
          <Prismatic number="7" a ="881.6" d ="480" minrange = "0" maxrange ="10000" maxspeed ="1890"/>
        </Joints>
      </Track>

    </Mechanisms>
    <IO>
      <DO names="1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16"/>
      <DI names="1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16"/>
      <AO names="1,2,3,4"/>
      <AI names="1,2,3,4"/>
    </IO></RobotCell>

boundlessmaking, 2019-05-05:
Now it works, thanks a lot. I will have to rethink my whole setup to figure out if an external axis is the right way to go or if I have to use other means to get the robot to move like I need it to. Thank you for your quick and helpful response.

closed 2019-05-05 by visose