visose/Robots #17: 7th version

metsah, 2019-05-12:
Hello,
I'm using Rhino 5 and the last Grasshopper on it. The last version I could use is the 5th is it possible to port your last versions also to rhino 5?

I really love by the way your plugin. I use it with a abb 1600-145 and it works pretty good of course with the right settings. I'm sure you will add in the future more features keep going :)
Best

visose, 2019-05-12:
> I'm sure you will add in the future more features keep going :)

Either I spend time on new features or in maintaining backwards compatibility, you can't ask for both :stuck_out_tongue: (although lately it's neither)

What's holding you to Rhino 5? Rhino 6 has been out for a long time now.

metsah, 2019-05-12:
Well, I understand you :) The problem is I'm living in turkey (€ is not that stable...) and I'm an educator. So no really money to update Rhino 6 :) ...

visose, 2019-05-12:
You can purchase a much cheaper educational version if you're teaching, did you check it out?

metsah, 2019-05-12:
Yes I know...And It's not just the money. I'm also a reseller of Rhino. And one of the biggest promoter here in Turkey for Rhino. And in my opinion there is not a major difference between rhino 5 and 6. I mean that I was waiting for more functions, which were implemented in the WIP versions... So in my opinion Rhino 6 is just a boosted Rhino 5 thats it; for that reason I'm waiting also for major updates or it looks like that I will wait until 7 is out which can be years :D

visose, 2019-05-12:
Being able to use the latest plugins is worth the upgrade.

metsah, 2019-05-12:
Functions like a good working Cycles, Mesh2, SubD, which will make a huge impact. The next thing will be also maybe an implementation of evee... And yes new plugins of course but the most are working better on rhino 5 for now...

closed 2019-05-15 by visose