visose/Robots #18: Running on old kuka robots

metsah, 2019-05-28:
Hello,

I tried to run some movements on an old 2003 Kuka 150 with a win95 based vxWorks. Is there a way to modify the output accordingly. Most of the time it has problems with the linking of the .dat files. Is it possible to somehow ''flatten'' the Code to one .src file and implementing the old declarations?

Best M.Ş.

visose, 2019-05-29:
What type of controller is it? I've tested the plugin with as old as KRC2 and it works without any changes. You might have to split the programs more often due to memory limitations.

 (As far as I know, modern KUKA robots still use VxWorks as their RTOS, except the iiwa)

metsah, 2019-05-30:
I'm not sure which version it is (I'm not next to the robot now) but it has a KRC2. The code that I send to the robot is just 400 Lines long. I'm testing it also with kuka/prc and here it works (in the settings you can click and enable a ''make compatible for older robots'' option).

visose, 2019-05-31:
I'd need more info to understand the issue. Variables in the .dat files are explosed as global variables, so make sure you don't have multiple of them with the same variables. To avoid conflicts, delete any other programs from the controller and have only the program you're currently running.

closed 2019-08-12 by visose