visose/Robots #20: Running on old kuka

metsah, 2019-08-21:
Hello,
I'm again on the old kuka 150. When I load the files to kuka one .dat and two .src files it generates errors:
on the .dat file - 2324 - dual decleration and one name not declared 2135
on the .src files it says linking errors, Syntax error in module

When I'm working with kukaprc it generates me just one file. Maybe this happens because of that linking problem. Is it possible to combine these files into one file or are there other solutions?

Thanks,
MS

visose, 2019-08-21:
Hi, can you attach a minimal grasshopper file with the issue and screenshot of the errors.

Delete any old src and dat files from previous progams just in case. Make sure you didn't name anything (tools, programs) with spaces or werid characters. 

The advantage of generating a dat file is that you can change those values while the program is running, making it more convinient to adjust things and faster to find the right values.

The reason it generates two src files is because the robot controller can't load large programs, they have to be split into subprograms, so it generates one main program plus any additional subprograms to load, by default one.

metsah, 2019-08-22:
Hi,
The kuka pendant shows me a folder called R1 in this folder it had some production files, which I deleted. But there are also some other folders like mada program etc... should I delete also the production files inside those folders?? There are some folders with some templates for arc welding etc.

I tried also to change the names of the variables, suddenly .dat and the short .scr file compiled but the main routine shows me a strange linking error on line -1 and column -1.
 
But when I use for instance kukaprc the robot does what it should :( .

I think your plugin is very good by the way I use it for another robot an abb irb1600-145 and it works  very good :) . The only thing that I'm missing is a good documentation a .pdf file with some examples maybe. Because I'm struggling to understand/program different coordinate systems.

Next week I will be again on the kuka and I will send you the photos from the pendant,
Thank you for your time,

visose, 2019-08-23:
I meant only the files that you yourself uploaded to the controller.
Can you attach a minimal Grasshopper file that shows the issue?

I'm sure it's not an issue with the robot itself, the most likely cause is user error.
It's possible the way this plugin works makes it more prone to user error than kukaprc. 
I'm still betting the issue is wrong characters in variables names, but can't really tell until I can check the file.

metsah, 2019-08-25:
Hi,
Here are the production files...
[freze9_kuka.zip](https://github.com/visose/Robots/files/3538112/freze9_kuka.zip)
and these are the rhino+
[Kuka_Robot.zip](https://github.com/visose/Robots/files/3538129/Kuka_Robot.zip)
grasshopper files

visose, 2019-08-26:
Your file name is 27 characters,  KRC2/KRC3 max file name length is 24 characters. This is probably the issue.
If you name it "freze9" it should probably be fine (including 'kuka' in the file name seems unnecessary 😛)

Note that the plugin appends some additional text to the file name of each subprogram (which has to be taken into account). This includes the name of the robot which defaults to "T_ROB1". This is to distinguish it from other robots in the case the controller controls multiple robots. It also appends numbers with some leading zeroes 001, 002, etc, for each subprogram.

visose, 2019-11-28:
I assume the issue was the name length and can be considered resolved.

closed 2019-11-28 by visose