visose/Robots #21: Universal Robot Speed Parameter

bastiannn, 2019-08-23:
Hi Vicente,

it seems that the robot runs slower when sending a program than it could possible run when it is programmed with the teach panel. 

Technical they should run at 1.000mm/s
In the teach panel it even says up to 5.000mm/s

How can I run the robots through Robots at maximum speed?

visose, 2019-08-23:
Are both programs identical? 
Could you attach a simplified Grasshopper file with the issue?

bastiannn, 2019-08-23:
I only put the load robot system component on the canvas. 

Dipl.-Ing. (FH) Bastian Wibranek, MA
mobil: +49 175 5244389 |
mail: wibranek@dg.tu-darmstadt.de  | http://www.dg.tu-darmstadt.de | https://www.facebook.com/DigitalDesignUnit |

Technische Universität Darmstadt
Fachbereich Architektur
Digitales Gestalten (Raum 333)
El-Lissitzky-Str.1
64287 Darmstadt

visose, 2019-08-23:
I mean, is the Grasshopper program that you are stream to the controller identical to the program you created using the teach pendant.
Having targets too close to each other, or not setting up the correct payload and center of mass might limit the robot's acceleration.

closed 2019-11-28 by visose