visose/Robots #22: DKP-400 base moves to rhino origin in simulation

evo85210, 2019-09-02:
I've download the newest version (8th/0.1.0) from the wiki download page
and I've noticed that the base for the DKP400 maves to rhino origin during simulation.
I'm using the bartlett xml+3dm (with modifications to match my setup)
screenshot of problem:
[image]

Have I mis-configured something? or is this a bug?

visose, 2019-09-02:
That's not supposed to happen. I'll have a look a bit later today. Could you attach or send me your 3dm/xml combo files for your robot.

visose, 2019-09-02:
I found the issue. Could you download and replace the 8th version and test if it works for you?

visose, 2019-09-02:
If you find it's still not working as expecting you can reopen the issue.

closed 2019-09-02 by visose

evo85210, 2019-09-02:
Confim fixed, thanks!