visose/Robots #28: Error loading robot system

baharmon, 2020-01-29:
Hi, I'm getting this error when trying to load robot system:

> 1. Solution exception:The 'Base' start tag on line 17 position 9 does not match the end tag of 'Positioner'. Line 24, position 9.

I believe I followed the installation instructions correctly. For installing the robot libraries, I tried downloading the repo and copying them. After reading issues 4 and 15, I also tried downloading the xml from raw view. Any ideas? 

Thanks, Brendan

visose, 2020-01-30:
Hi Brendan, my guess is that you have a malformed XML.
Check that the "base" tag on line 17 is self closing, with a forward slash at the end, like this:
`<Base x="0.000" y="0.000" z="0.000" q1="1.000" q2="0.000" q3="0.000" q4="0.000"/>`
Could you attach it here?

baharmon, 2020-01-30:
I've tried a new document and your simple example.gh using the Bartlett library. The error message is the same before I set a robot name, when I set robot name, or when I open the example. The line numbers in the error don't match the code in bartlett.xml.

[bartlett.zip](https://github.com/visose/Robots/files/4134166/bartlett.zip)

[image]
[image]

baharmon, 2020-01-30:
Ah, I forgot to unblock the rhino model bartlett.3dm. Now it works. Thank you for the help.

closed 2020-01-30 by baharmon

visose, 2020-01-30:
Great 👍

cfboyd, 2020-10-11:
Hi I'm having the same error but I did actually unblock bartlett.3dm. I've also downloaded grymsdyke to see if the bartlett file was the issue, and I can see the grymsdyke robots in the dropdown menu but i still get that error with the bartlett ones, the grymsdyke ones, and the default robots.

cfboyd, 2020-10-11:
I have also downloaded the xml files from raw and believe I've followed the installation instructions

cfboyd, 2020-10-12:
Ignore this. The issue ended up being that grasshopper was looking in my user/onedrive/documents rather than user/documents for the Robots folder