visose/Robots #32: Simulation plane by plane

petrasvestartas, 2020-05-21:
Is it possible to simulation robot position by iterating over plane by plane?

When I am using Program Simulation component with normalized values from 0 to 1. It is very hard check the target position. Ideally I would like to have a slider from 0.00 to n where n is the number of planes and each integer 1.00 or 2.00 would give a robot pose at the given plane.

visose, 2020-05-21:
Yes, you can use the 'Deconstruct program targets' component. It outputs different parameters that are calculated for each target when creating the program. You connect it to the Program output of the 'create program' component.

You can then use a integer slider and list item to extract parameters one by one. You could either extract the joints and then use the Kinematics component to visualize the robot, or you could use the partial sum of the 'delta time' and feed this as the time value of the simulation component.

petrasvestartas, 2020-05-22:
Thanks it works, I added short snippet:
```

private void RunScript(List<double> T, double t, ref object A, ref object B)
  {

    //Constrain to end
    double p = Math.Max(0, Math.Min(T.Count - 1, t));
    Print(p.ToString());

    //Mass addition
    double[] massAddition = new double[T.Count];
    massAddition[0] = T[0];

    double sum = T[0];
    for(int i = 1; i < T.Count;i++){
      sum += T[i];
      massAddition[i] = sum;

    }

    //Interpolate
    int id = (int) Math.Floor(p);
    Interval interval = new Interval(massAddition[id], massAddition[(int) Math.Min((T.Count - 1), id + 1)]);
    double Time = interval.ParameterAt(t % 1);

    A = Time;
    B = id;
  }
```

closed 2020-05-22 by visose