visose/Robots #34: Will not load = Grasshopper Breakpoint issue

MarceMoulton, 2020-06-25:
Good Morning,

Robots will not load because of a problem listed below.
Can anyone point me in a good direction to solve please?

Thank you

"Grasshopper Breakpoint
The RobotsGH Component Library was buid against a different major version of the Grasshopper SDK.
Local SDK = 1.0.0.20
Reference SDK = 6.11.1834420090

visose, 2020-06-25:
Check your Rhino 6 version, it should be 6.11 (as in the number above) or higher.
To check the version number go to Help > About Rhinoceros.

MarceMoulton, 2020-06-26:
Okay will do so.
Thank you!

I was attempting to run in Rhino 5 as the Wiki says.
Prerequisites
    Windows 7 or above
    .NET Framework 4.0
    Rhinoceros 3D 5 SR13 64 bit
    Grasshopper 0.9.0076

MarceMoulton, 2020-06-26:
Hello,

Need a bit more help please.

Running Version 6 SR26
(6.26.20147.6511, 5/26/2020)

In R6 no Grasshopper Breakpoint error.
But Robots does not showup in Grasshopper.

Unblocked Robots.dll and Robots.gha are in:
C:\Users\RatCat\AppData\Roaming\Grasshopper\Libraries

All downloaded libraries are in:
C:\Users\RatCat\Documents\Robots

Any idea what I might change?

Thanks!

visose, 2020-06-26:
All seems correct, try dragging robots.gh to the Grasshopper canvas and see if you get any error messages in the Rhino console.

visose, 2020-06-26:
> Okay will do so.
> Thank you!
> 
> I was attempting to run in Rhino 5 as the Wiki says.
> Prerequisites
> Windows 7 or above
> .NET Framework 4.0
> Rhinoceros 3D 5 SR13 64 bit
> Grasshopper 0.9.0076

Thanks, I updated the Wiki with the current prerequisites.

MarceMoulton, 2020-06-26:
Fantastic!
Dragging did the trick - seems to be working now.
Will test and confirm in a bit.

Thank you.

MarceMoulton, 2020-06-26:
Good Evening,

All works well except I must drag and drop Robots.gha into Grasshopper each time I wish to use.
For some reason will not stay in memory.

Also, is there a way to import a slicer file (gcode)?

Thanks tons!

MarceMoulton, 2020-06-26:
Where or how can I find more information about components?
Looks like custom code may work - wondering?

Thx

visose, 2020-06-30:
> All works well except I must drag and drop Robots.gha into Grasshopper each time I wish to use.
> For some reason will not stay in memory.

That's very strange, you shouldn't have to drag it every time.
 
> Also, is there a way to import a slicer file (gcode)?

There's something for G-code in the other pinned repo called Extensions, see:
https://github.com/visose/Extensions/blob/master/Extensions/Model/Toolpaths/Milling/GCodeToolpath.cs

visose, 2020-06-30:
> Where or how can I find more information about components?

There's some info in the Wiki here. Besides this, some people have made tutorials, but might be hard to google due to the original and distinctive name of the plugin.

> Looks like custom code may work - wondering?

The 'Custom Code' component is used when you want to manually modify in Grasshopper the text code that is generated by the 'Create Program' component. Usually using components such as 'text replace', or inserting custom text snippets. You would feed this changed code to the custom code component to be able to use other components such as 'Save program' or 'Remote' with your edited code.

MarceMoulton, 2020-06-30:
Fantastic!

Reviewing now - will let you know how it goes.

Cheers,

closed 2020-06-30 by MarceMoulton