visose/Robots #38: Load robot !

seghier, 2020-08-19:
Hello , i try to load robot but no list available
How i can choose the name?
And can't open any example file , always error
IO Read error: Unknown data type encountered.

visose, 2020-08-19:
If there's no list of robots it means you didn't properly add a Robots folder with a robot library (the .3dm and .xml files) in your Documents folder.
Did you follow the install instructions in the Wiki?
When downloading an .xml from GitHub, make sure you don't accidentally download the webpage instead. Try downloading this whole repo as a zip file and get it from there.

seghier, 2020-08-19:
Thanks
Yes problem was with xml file.
The second problem still exist; i can't open example files

visose, 2020-08-20:
You probably didn't download it correctly. But you should provide more info to be able to help.
Upload one particular file that you have the issue. Is it a Grasshopper file? How are you opening it exactly?

seghier, 2020-08-20:
I download the files from this link
https://github.com/visose/Robots/tree/master/Documentation/Examples
And this is one of them from my pc, i never have problem with gh files.
https://www.dropbox.com/s/ajv6dx70b7lm5wh/simple%20example.rar?dl=0

visose, 2020-08-20:
As suspected, you downloaded the website, rather than the file. You can see this by opening the file in a text editor.
When you downloaded the whole repo as a zip, the examples should've downloaded correctly, use those ones.

closed 2020-08-20 by visose

seghier, 2020-08-20:
Thank you, that's weird i never have this problem before when i download gh files from github.