visose/Robots #41: [help request] Staubli robot

eduardsik, 2020-09-27:
Dear Vincente, 
May I ask for your help, to fix our robot Staubli XT60 in the bartlett robot repo? I put it in the xml and 3d file, but it is falling apart. 
Here is the dropbox folder: https://www.dropbox.com/sh/4zxnwty97zgniyu/AAByzYfS1AUnCoC9EWcQlaEAa?dl=0

Thank you in advance.

vinceHuyghe, 2020-09-29:
Hi Eduard,

The values in your XML file are not correct. This is why when you move the
robot it comes apart.
If you look at the data sheet for your robot and update the a and d values
in the xml it should fix the problem. (you can look at the datasheet for
other staubli robots already in the library to see what dimension goes
where)
Hope that helps.

Best,
Vincent

eduardsik, 2020-09-29:
Hello Vincent,

Thank you for the suggestion. I thought the problem is in the 3d model. I took the dimensions based on the other staublis. But I'll check again. I would like some help on the meaning of a and d.

eduardsik, 2020-09-30:
Solved

closed 2020-09-30 by eduardsik

vinceHuyghe, 2020-10-01:
Hi Eduard,

The A and D values refer to DH parameters.
https://en.wikipedia.org/wiki/Denavit%E2%80%93Hartenberg_parameters

Best,
V