visose/Robots #42: ABB IRB 6700 150/3.2

Anenus, 2020-10-28:
Hello, will ABB IRB 6700 150 / 3.2 be available with this plugin?

v-xup6, 2020-11-06:
You can simply add your own robot by adding the model to the 3dm file and add its parameters to the xml file.

closed 2020-11-06 by Anenus

Anenus, 2020-11-06:
Done, thank you very much !

I'm trying to make a simple pick and place script with your plugin (only one box), but I'm struggling with the toolpath.
I created the Tool, the robot, the box and the planes (origin and target for the TCP). However, your example is very complex and doesn't really work (I wasn't able to find the commands to add/delete)

Are there any simpler examples? (for only one box to pick and place)

Thank you very much.

Regards,
Walid