visose/Robots #43: Bug with joint target generation

gabzolina, 2020-11-04:
Hi there, we noticed a bug in the generation of joint targets for ABB robots.
Example:
Input panel with 6 axis values in degrees  (0,10,30,0,-40,0)
Transform to radians and concatenate in a string separated by commas
create joint target
the rapid code generated is (0,80,-30,0,-40,0)

We have tried with several x values for axis 2 and y value for axis 3, the rapid code returned is always 90-x for axis 2 and -y for axis 3.

gabzolina, 2020-11-04:
[robotsBug.zip](https://github.com/visose/Robots/files/5489046/robotsBug.zip)

v-xup6, 2020-11-06:
I've also noticed this for KUKA robots, I assume it's simply because the kinematics model of robots does not match the 0 position and joint direction of the physical robot.
It's not a big deal, there's a Degree to Radians component in robots that'll convert the joint values of the physical to match Robots' kinematics model. see attached file.
[robotsBugSolution.zip](https://github.com/visose/Robots/files/5498413/robotsBugSolution.zip)

visose, 2020-11-14:
Hi, this is not a bug. You can find more info probably on closed issues opened by other people that also thought it was a bug.
I understand this violates the principle of 'least astonishment' so people get confused, but there's a good reason behind it.

closed 2020-11-14 by visose