visose/Robots #44: Help request (Staubli TX60)

eduardsik, 2020-11-05:
Hello,

I would like to ask for help with Staubli XT60, to add to the library. I modified the bartlett library and add the robot to xml, and 3d model, but can not fix to not falling apart, even if using the DH parameters. 
May I ask you for add the robot correctly? 
Here are the necessary files (3d model, xml, DH parameters): https://www.dropbox.com/sh/9ziq0qys1qapfl9/AAC4jETHgbto-r9fl0a2YIYma?dl=0
Thank you very mutch!

v-xup6, 2020-11-06:
I think the problem is with the plugin itself?
TX60 is slightly special in that not all joints are on the XZ plane, I think this is what's causing problems.
[image]

I've also noticed that the plugin does not read the d value for Revolute number 2 and number 3, you can change it to some absurd vlaue like 5000 and there'd be no difference.

Will need Vicente or Vincent to confirm this tho, just pointing out my observations.

eduardsik, 2020-11-15:
@v-xup6 Thanks for you observation. I saw this shift in design on other staubli robots and it is involved in xml. I think somewhere else is the problem, but I can't find it.

visose, 2020-11-15:
Hi, I recommend you get a view with dimensions like above for one of the existing Staubli robots in the Bartlett library, see what dimensions the numbers correspond to, and then replace them with the TX60 numbers of the same dimension.

eduardsik, 2020-11-16:
I did this as you suggested and also calculated the DH parameters for the dimensions. As a result, the robot is much better assembled, but it always breaks down.  That's why I'm looking for your help.

visose, 2020-11-16:
Strange, you shouldn't be calculating any dimensions, just copying the exact number from the drawing.

eduardsik, 2020-11-16:
i checked once again and seems the data is good in xml. Could you please take a look at it? Maybe can ask you to create a sample sketch in grasshopper with TX60 to test it out and send to me as a reference? It would help a lot.

visose, 2020-11-16:
Ok, I'll try to have a look in the evening. Just giving you a chance first, since it always feels better to figure out the issue on your own.

eduardsik, 2020-11-16:
Thanks, I'll check one more time.

eduardsik, 2020-11-19:
Did you manage to take a look at it?

eduardsik, 2020-12-18:
Sorry but no success on my side. Could you manage to look at the files please?

visose, 2021-01-17:
@eduardsik  I had a look at this and @v-xup6  is correct, that offset is not supported in the kinematics used by Robots (the d value of the second axis is not used). I don't think I will have time to work on adding this anytime soon, but would be a great if anyone else is willing to try.

visose, 2021-01-17:
I will close this issue, the conclusion being, not a supported feature yet.

closed 2021-01-17 by visose