visose/Robots #46: AR3 Robot

pappurvsa, 2021-01-06:
Hi Vicente,

Hope you are doing well. 

I built an Open Source [AR3 Robot ](https://www.anninrobotics.com/)over the break out of my personal interest. Still work in progress, but works. 
https://user-images.githubusercontent.com/8084742/103826424-b77d5100-506e-11eb-89fb-f56b90e84451.mp4

I have been trying to reuse the robot's plugin for the simulation by modifying the 3dm file and XML file.
[AR3.zip](https://github.com/visose/Robots/files/5778511/AR3.zip)

But the simulation seems off. 

The robot uses a left-hand coordinate system as compared to the RHCS in Rhino and the DH parameters for 4th axis are negative which my brains do not understand. Please find the DH parameters attached.
[image]

Please let me know if you could help. Any help/ suggestions/ advise would also be really helpful.

I'm working on a separate post-processor with my minimal knowledge in Python. Hope it works as AR3 uses pickle which I just found out today. Will share once it is a bit it works.  

Many thanks.

Kind regards,
Pradeep Devadass

visose, 2021-01-17:
Hi Pradeep, hope you are well! I tested the xml/3dm you attached and it seems to work fine, the only thing I changed is the name of the robot cell from KUKA to AR3. Can you describe the issue in more detail?

pappurvsa, 2021-01-17:
Many thanks for your response.

The DH parameter for d4 is -222.63 while I have entered it +222.63 in the XML. Similarly, for the d6. If I enter the negative values the robot model goes crazy. 
Does this make my robot Axis 4 flipped. Please find the image attached for J1=-100, J2=-120, J3=120, J4=60, J5=90, J6=90.
[image]
[image]

Is it possible to change the **'alpha' and 'theta'** DH parameters for a robot configuration, so I can make the exact kinematic configuration of the AR3 robot?

Also, the robot uses a left-handed system while Rhino uses a right-handed system. Would that make difference in the simulation?

Please let me know.

visose, 2021-01-17:
In Robots, the joint rotations used internally are converted differently depending on the robot manufacturer. I assume the same will have to be done with this robot. Besides converting from radians to degrees (if your robot requires degrees), it can also be things like reversing the direction, adding 180 degrees, etc. and this might be different on each axis.

visose, 2021-01-21:
Hi @pappurvsa let me know if the fix was to adjust the conversion of joint angles, to close the issue.

visose, 2021-02-17:
Hi @pappurvsa I will close the issue assuming it was related to my last comment, but you can open it again if not.

closed 2021-02-17 by visose

pappurvsa, 2021-03-23:
Hi Vicente,

Many thanks for your support. Apologies for the delay in reply as I had to fix and test the script with that robot. I did manage to fix the issue to a certain extend. I used 'Expand Objects' component from Speckle Library (Download link [https://we.tl/t-HUzyqy8VlX](url)) to extract information from the Target Parameters (goo object). I wrote a post-processor in  Python.

Following is an example I tested a simple Pick and Place experiment using this method.

[https://www.linkedin.com/posts/pradeepdevadass_architecture-design-construction-activity-6778201411911057408-VqJh
](url)

The problem I'm still having is converting the rotation angle around X and Y from RHS of the rhino environment to LHS which the robot uses. Is that something you could help with?

Please find the files attached.
[AR3.zip](https://github.com/visose/Robots/files/6192371/AR3.zip)

Many thanks again.