visose/Robots #47: Invalid Mesh

W1711, 2021-01-24:
Hello Vicente,

I was trying the robots plugin for some pick & place scripts. But every time I lose a mesh when I run the script. This varies between robot meshes and gripper meshes.

Example for the gripper, in its closed state, its mesh is correct but in its open state, the mesh is invalid.
For the robotic arm I use the Bartlett-UR10, and going through the script, the meshes of the arm axis get lost alternately.

This is due to what?

[image]
[image]

visose, 2021-01-24:
Hi, are you using the latest Rhino 7? Someone else contacted me with the exact same bug using Rhino 7. I can have a look if you attach a minimal rhino and grasshopper file that can reproduce the bug.
As a workaround you can manually orient the tool mesh to each target position using the Orient component and the resulting Planes for each target (from WorldXY plane to target 6th axis plane or from tool TCP plane to target TCP plane).

W1711, 2021-01-24:
Yes I'm using Rhino 7 right now. I also think that it's about the Rhino version because I didn't have those kinds of problems with the Rhino6 version.
Here is a simple grasshopper script & the rhino file with the tools:

[https://drive.google.com/file/d/15YFAN93U5syQl0xmOqi1c4ja58JMiwvS/view?usp=sharing](url)
[MeshBug.zip](https://github.com/visose/Robots/files/5863386/MeshBug.zip)

vinceHuyghe, 2021-01-25:
Have a look at the discussion in this thread it fixed the issue for me
https://discourse.mcneel.com/t/orient-mesh-creates-invalid-mesh/104082/2

W1711, 2021-01-25:
It is true that rebuilding the mesh with Pufferfish can resolve the problem. For the tool it works fine, but for the robotic arm it would take a large amount of computational time, so it won't be efficient for a complex script (even for a simple one) :(

visose, 2021-01-25:
Haven't had a look yet, but from the post above it looks like it's likely an issue with the original mesh. The mesh is not considered valid from the beginning, but this only gets picked up after transforming the mesh, and Grasshopper decides not to display meshes flagged as invalid. The solution would be to fix the original mesh. If it's a robot joint, by repairing the mesh in the .3dm of the robot library.

W1711, 2021-01-28:
Solved, the meshes has to be fixed from the beginning. Thanks!

closed 2021-01-28 by visose

UnfoldAntwerp, 2021-08-17:
@W1711 how did you fix this? I'm loading the UR10 from KULeuven and this one has the same issues. Looks like the other UR10s (Bartlett etc) are the same mesh.