visose/Robots #48: UR10 With Track Problem

m4walker, 2021-01-26:
Hello,

I have recently tried to construct a custom robot cell (UR10 and linear external axis), and have had some problems with loading the robot system within grasshopper. Basically I had taken an existing 3DM and XML file (AAU_UR10) and added my own track geometry while also integrating the necessary code within the XML to account for this track geometry in the 3DM. Although no errors are present explicitly, the problem occurs when trying to load the custom library. Only the robot arm will load in when the library is referenced while the track geometry is not present. I have tried this again and again, using different names and manufacturers to test the validity of the script. I found that if an ABB or KUKA robot is used, and the track manufacturer is also declared as ABB or KUKA, then the custom library loads with no problem. This problem only seems to happen when UR is declared as the manufacturer. I have attached the custom library here (SAPL-UR10e-OnTrack), along with the screenshot showing the problem. 
I am not sure if this is a bug in the program or if I am doing something wrong, but any pointers from here would be much appreciated.

Thanks!

[SAPL-UR10e-OnTrack.zip](https://github.com/visose/Robots/files/5870778/SAPL-UR10e-OnTrack.zip)
[image]
[image]

visose, 2021-01-28:
Hi, I might be wrong and this might not be the case anymore, but when I added UR to the plugin, the controller didn't have support for external axes. Therefore, it is not possible (as it wouldn't make sense) to add external axes to a UR robot in the plugin either. This only makes sense if the kinematics of all the axes is handled by the same controller. Doing a quick google I see there are some solutions for a linear axis for UR, but it's not clear if the kinematics is integrated.

If this is your case (if you move the track, the robot motion is not affected), you can just use the move and orient components to simulate the track independently of the robot.

m4walker, 2021-01-28:
I see. That makes sense. We are using a track manufactured by a different manufacturer (Vention), where we are trying to use their URCap commands within the final script that we send to the robot. So in a way it is handled through the same controller, but it is done so through an integration with the third party software, not something done explicitly with UR alone. I see with ABB or KUKA that the external axis is integrated directly with the commands, so that explains why it performs differently.

visose, 2021-01-28:
The main thing is if the kinematics is integrated, if you tell the robot to move to a particular point and move the track at the same time, the robot will still reach that point. 
If a robot controller triggers an IO command that starts another mechanism, like a conveyor belt, sure, the conveyor belt is being turned on by the robot controller, but the motor is not really controlled by it. I assume this is the case but with more fancy commands to control the track, but still not integrated kinematics (I could be wrong).

closed 2021-01-30 by visose

dreamhuang, 2021-07-13:
@m4walker  Hello, we plan to also use Vention system as 7th axis for UR. What's the result of your attempt, about Synchronous the Vention and UR? Thanks in advance for your sharing.