visose/Robots #49: URS file not supported?

architektpetr, 2021-01-30:
Hello. 

I just started using Robots instead of HAL3000. This plugin is well done and looks great. May I ask for help with four issues? (I use UR10): 

1. Robots generates .URS file which is not recognized by UR PolyScope. The file extension needs to be .script ... Is there a way how to change this automatically? Also, there is still the possibility that I do something wrong. 
2. The script won't run unless I call the function at the end (manual edit of .urs file). For example: 

def Program():
movej(...)
movej(...)
**Program()** _<<< this needs to be added at the end_

3. Also, my remote access to the robot doesn't work. But that might be just absence of my knowledges. Is that just that easy - plug robot in, set IP address and "play" the program? Or does it require some additional setup in PolyScope? 
4. Is it possible to edit acceleration? 

Thanks! 
Petr

visose, 2021-01-31:
Hi, once you get the text output from the Create Program component you can use standard Grasshopper components to make those changes, like add Program() line to the end of save it with any file name and extension.

You must put the IP address of the robot in the Grasshopper component. There shouldn't be any additional step required in the UR pendant, except making sure the network settings is enabled and the robot is configured with a fixed IP.

The Speed parameter does have acceleration, but it's not exposed as an input. You can modify the value using a scripting component.

architektpetr, 2021-02-01:
Thanks for comments! One last thing: Value list doesn't work automatically, but I have to input string from panel or so (for example "Batlett_UR10". The xml file is copied and unblocked as you described in the manual. 

Here is how I modified your component, so it exports .script file with proper syntax: 
(I was not able to upload .gh file, so it's [linked from my cloud](https://drive.google.com/file/d/1MOpv3jyYT4I10jvFAPrLT-AYHEfqwIOP/view?usp=sharing))

[image]

visose, 2021-02-05:
You have to place a new Create RobotSystem component for the value list to populate with the list of robots.

visose, 2021-02-17:
Hi, I will close the issue assuming this worked, you can reopen if not.

closed 2021-02-17 by visose

architektpetr, 2021-03-14:
Hello

I tested the IP connection just today and it works well. Thanks for
suggestions.
There's still some trouble with the definition of the robot - the value
list doesn't work, I have to write "Bartlett-UR10" to a panel and plug it
in. But the plugin works fine. Just sometimes there are error messages
without any other hint of what is wrong.
Also, is there a way to set "time" input for PTP movement?

Best regards,

*Petr Vacek*
Architekt | Computational Designer | Maker

<https://www.linkedin.com/in/petr-vacek/>
<https://www.instagram.com/_petrvacek/>
<https://cz.pinterest.com/vacekpetr/boards/>
<https://www.facebook.com/petr.vacek.7739/>

petrvacek.com

+420 732 92 82 82

***@***.***

Prague, Czech Republic <https://goo.gl/maps/uYaR4pUhk92UXv2C9>

robin-gdwl, 2021-03-14:
The time input is not exposed in the standard grasshopper component same as the acceleration input. 
You need a script component and access the Speed-Parameter there and set a time or acceleration value. 
I recently did the same thing as documented here:

https://github.com/visose/Robots/issues/51#issuecomment-775253135