visose/Robots #50: Customizing a line in a program

gkirdeikis, 2021-02-05:
Hi there,

Since we have quite a few modules that are constantly loaded into our ABB robot PROC Main() command always complains and I need to get into it and manually change Main to something like MyCustomMain() .
I was wondering if it's possible to do that in Grasshopper somewhere between Create Program and Save Program components.
TL;DR - is it possible to "unpack" the program, change it and then "package" it for the SaveProgram component?

visose, 2021-02-05:
Hi, yes. You can modify the code output of the Create Program component with standard grasshopper components, like using the Replace Text component to replace Main() with MyCustomMain(). Then connect this code and the Program output to the Custom Code component, and from this to the Save Program component.

gkirdeikis, 2021-02-05:
Understood, thank you for the swift response!

closed 2021-02-05 by visose