visose/Robots #51: [FeatureRequest] Add the possibility to specify a time parameter in the speed component

robin-gdwl, 2021-02-08:
Some Robot Languages (like UR) allow for specifying the time a movement should take. 
UR implements this with the `t=xx` parameter in most move fuctions. 
I saw that this functionality is already available in the `RobotCellUR` but I may be mistaken: 
https://github.com/visose/Robots/blob/9fc97b7ced60f321af310d68ee2cf1ea4b838ecd/Robots/RobotCellUR.cs#L342-L343  

would it be possible to expose a `time` input in the speed component or would that be confusing/break things for robots that don't use this functionality?

visose, 2021-02-08:
You can create or modify a Speed parameter using a scripting component. You can there set variables not exposed as inputs such as Time and Acceleration

robin-gdwl, 2021-02-08:
I will try that! Thank you for the quick response.

closed 2021-02-08 by visose

robin-gdwl, 2021-02-08:
Could you give a quick overview how I can get access to the Robots-.NET library in the scripting component? I am having a hard time configuring the right namespaces etc.

Do you have an example script I could see as reference?

robin-gdwl, 2021-02-08:
I made it work, with the help from this previous answer of yours: https://github.com/visose/Robots/issues/19#issuecomment-520556168 
Other examples would still be appreciated.

robin-gdwl, 2021-02-08:
For Reference: this is the code I used to create a speed parameter with a time value:
Written in C#. I did not need to specify `using Robots;` after adding the Assembly by right clicking on the c# scripting component. 

```cs
//using Robots;
//namespace Robots;

var inputspeed = new Speed() ;
inputspeed.Time = time;

Robotspeed = inputspeed;
Print(inputspeed.Time.ToString());
```