visose/Robots #54: How to learn to control robot once you buy it?

petrasvestartas, 2021-04-07:
Hi,

I have a question not completely related to robots plugin.

I would like to ask if there are any manuals for this scenario:

a) user buys abb robot and gets delivered to his/her workspace
b) user installs Robots plugin on rhino grasshopper

Since a company that installs robots does not know anything about Robots plugin nor Rhino how a user can establish a physical link between robot and rhino? Is there any proper manuals? 

I do not even know if connection works via ethernet cable... and basic steps.

robin-gdwl, 2021-04-09:
Did you see the section for ABB robots in the wiki? 

```md
# abb
In the latest version you can use the Remote component directly from Grasshopper to upload, load and optionally run the program on the controller. You'll need to install the Robot Communication Runtime if you don't have RobotStudio installed.```

petrasvestartas, 2021-04-09:
Dear @robin-gdwl 

Thank you for reply.

The second - rather the actual connection (setting up a network, connecting ethernet cables ...) that you need help with.

robin-gdwl, 2021-04-09:
I see, I have not yet used ABB Robots so I cant help you  myself. 
It probably also depends on what robot and controller you use. 
This video should be a good starting point: (dont let the UR-Robot confuse you its about connecting to ABB)
https://www.youtube.com/watch?v=LiV7xgSIZKA 
Since Robots uses Robot Studio or  [ABB Robot communication Runtime ](https://forums.robotstudio.com/discussion/10677/robot-communication-runtime-6-06-01-sp1-is-available-for-download) to interface with the robot I would check if you can find a manual for those that details how to connect.

petrasvestartas, 2021-04-09:
Thank @robin-gdwl 

Is it also valid for both old and newer abb i.e. IRB 6400?

robin-gdwl, 2021-04-09:
> Is it also valid for both old and newer abb i.e. IRB 6400?

I'm not sure, I have only used robotstudio as a simulation not with a physical robot, but I guess this info should be found somewhere in the manuals for roboguide or the controller itself.

petrasvestartas, 2021-04-11:
Thanks for the help;)

closed 2021-04-11 by petrasvestartas

visose, 2021-04-11:
@robin-gdwl thanks for answering 👍🏻