visose/Robots #55: UR5 don't move if you have too many targets

vsardenberg, 2021-04-19:
Hi Robots people!

When I am using robots to send programs to my UR5 it only works if I have less than ~2500 targets. So, I've been splitting the programs, which is not ideal. Do you have a solution for it?

Best!

visose, 2021-04-20:
Only solution I think is to not use the stream program feature, copy the script to the controller, create a program using the pendant that runs the script (you might or not have to split it into multiple subprograms).

closed 2021-05-12 by visose

robin-gdwl, 2021-05-14:
@visose what is the bottleneck in the connection? a TCP-connection via LAN should have no problem sending a few thousand lines of code in a short time. 
Is the program checked by the robot pendant? Is the problem on the side of the Robot-Controller? or is there a way to speed it up on the GH/Robots side?

visose, 2021-05-14:
I am not exactly sure where the limit is. Usually very large programs have to be split into multiple subprograms, because when a program is loaded it has to fit in the memory of the controller. This happens with milling programs made of lots of targets or the very common practice in architecture school of using an industrial robot to slowly extrude thick plastic and calling it "3D printing".
Most robots have a way of automatically loading/unloading a sequence of sub-programs.

In this case this is probably different as the size is smaller. It could be a limit of the size of the data that can be sent to the UR in one go through via streaming. But haven't seen in the documentation anything about a limit. I think the best is to use the streaming function for tests and for the final program with lots of targets create a URP program and assign a script generated with Robots, the script can be sent to the controller by FTP or USB.