visose/Robots #56: Using a message from the UR5 Pendant to pause the programme

vsardenberg, 2021-04-19:
Hello, Robots people!

I am working on a pick and place project where I need to screw parts of an aggregation. We will use a 2 finger gripper to pick and place slats in specific positions and humans will screw them together. Because we don't know exactly how much time it will take, we would like to implement on the program an instruction to pause it and send a message box to the pendant asking if you want it to continue. Is it possible to do it?

Thanks a lot!

Victor

Yonahel, 2021-05-11:
Hello Victor,

Disclaimer - I don't work with the Robots team, but I have been experimenting with the plugin for a few weeks. 

There's a Wait DI command that comes with the plugin, that seems to have some of the functionality you're looking for. You could set up the program to do a Pulse DO, to tell your pendant it's reached that point in the program, then Wait DI. The message box would awaken with the Pulse DO, and then, on continue, send a digital out back to the UR controller which triggers the Wait DI. 

That should do what you're asking for, although I don't know if you know how to interface your pendant's message box with digital in and out signals. If not, this might not be what you're looking for.

robin-gdwl, 2021-05-13:
What robots are you using?
For UR-robots you cann add a `popup` command that blocks the robot until you press continue on the pendant.
Check the UR-script reference for more information.