visose/Robots #58: Some weird joint config problem

svitDa, 2021-05-12:
Hello, in my extrusion script, sometimes the robot does a flip of TCP. The path goes from one point with the same orientation, to the same one, ony the xy position is different, and he just decides to do a flip of the tool. Dont understand why. Also tried to restrain the movement range, it does not produce different solution, only "axes out of bounds" . Hilarious. Video here: https://youtu.be/mLg5ein-59o 
Any idea how to fix it? 
Joint or linear, still flippin. 
Tried different config, but that is pointless. 
Needs a different solver.

visose, 2021-05-12:
A few quick things:
- If you post a minimum reproducible example someone can have a look.
- Would be great to check the error output of the create program component to see what's the issue.
- The disappearing of the meshes in the video its likely related to a change in Rhino 7. There is a closed issue here about it.

Just a suggestion for the future: The video title, the use of the words hilarious and pointless, and confidently stating that it needs a different solver makes me categorize you as a smartass not worth helping. You might not be, but I don't know anything else about you except this comment.

svitDa, 2021-05-12:
Sorry if it sounded too smartassy. 
https://wetransfer.com/downloads/b039851a52fda0a0ed68ce479ff8414820210512134334/00ec12af080ce6dfbd9208be672e7f3720210512134359/b1f595
Here is a link to a file which shows this weird behavior.
It is a UR5e, but I am not sure if that is the issue. 
The error outputs are just that I tried to fix this by constraining the axes of the robot, limiting the axes, so that this flipping would cease to happen. 
Hope you did not get offended, its just that I am reaching out after weeks of banging my head against the wall, so I admit, I was frustrated. 
Hope you can point me in a direction, thanks.
Daniel

svitDa, 2021-05-13:
Well, here it goes: as always, the problem was somewhere between the chair and the keyboard. Meaning, I had the target planes simply wrong, the coordinates XY were pointing in the same direction for the whole path, and by adjusting the Y axe to point to the base of robot, all seems to line up great.

closed 2021-05-13 by svitDa

visose, 2021-05-13:
From the video it also seemed like a singularity issue, that's why it would be good to know the error message on the create program component. Reorienting the targets might have fixed this.