visose/Robots #62: Can't load robots on Windows after installing for Mac

laurend, 2021-07-19:
I have been using Robots on a virtual Windows machine (Rhino 6) for a few years now. 
Recently I started using Robots with a new installation of Rhino on the Mac side of my computer (Rhino 7). 
Now, when I try to load a robot in Windows (Rhino 6 or 7), the component is looking in the directory that corresponds with my Mac installation, which obviously doesn't exist. The folder of these files on the Windows side is still in-tact. 

I tried updating the robots plugin on the Windows side without any success. 

This happens both for Rhino 6 and Rhino 7 in Windows and occurs both for GH files that were created previously and new instances of the component. 

Is there a way to manually set this folder that Robots is looking for the robot definitions? Or perhaps there is another workaround?

[image]

robin-gdwl, 2021-07-20:
What program do you use for the virtual machine ?

laurend, 2021-07-20:
I'm using Parallels. I just tried disabling the folder sharing and uninstalling/reinstalling robots. Then I got this error:  
[image]

Turns out I had some hidden files in the Robots folder that I think came from making changes to some of the robot definitions. 

I removed those files and now it works. Thanks @robin-gdwl!

closed 2021-07-20 by laurend