visose/Robots #129: Sanitise string inputs for tool/program names

robin-gdwl, 2022-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.0.1 afaik also happens on 1.0.4

### Version

Rhino 7 on Windows

### Current Behavior

The inputs to strings like `Tool Name` and other similar Variables are not sanitised. 
If a Space, punctuation or any other forbidden letter / symbol is present it will be put into the final robot code as-is. 
This then leads to difficult to diagnose bugs and issues that are not obvious to non-programmers.
A tool name such as "2xGripper" might result in an error. This is of course different depending on the robot language. 

When trying to upload a program to a UR-Robot for example nothing happens, there is sadly no warning or error message either in Grasshopper or on the robot. 

Our students run into this problem all the time unfortunately. It would be great if this could become more tolerant to user input errors. 

### Expected Behavior

There could be multiple steps to prevent this. 
- The description for inputs such as `Tool Name` (and other strings that are pasted directly into the robot code at the end) should remind users, that not every letter is allowed and they should avoid certain letters or at least refer to the robot-language specification for guidance on variable names
- The `Create Program` component should give an **error** or at least a **warning** if a string is empty, multiline, starts with a number or includes punctuation
- The string input should be sanitised at some point and all special characters automatically discarded or exchanged
- The wiki could mention this user-error in a troublshooting guide

### Anything else?

Happens on both Win and Mac.

visose, 2022-01-16:
Good suggestion👍🏻

Each manufacturer has slightly different rules for names, but 99% of the time I see someone with this issue it's, in order:
1. Spaces
2. Leading numbers
3. Invalid characters

I'll try to find out/remember others, like the char limit in KRL, but these should take care of the most. Feel free to add a comment with other restrictions.

visose, 2022-01-17:
I've decided to use the same restrictions no matter the robot manufacturer.

Advantages:
- Changing the robot system to a different manufacturer will not require name changes.
- Can throw an error as soon as the named attribute is created, before creating the final program.
Disadvantages:
- Might be more restrictive than necessary in some cases.

The restrictions are:
- Must contain at least 1 character.
- Must contain 32 or fewer characters.
- Must start with a letter.
- Can only contain letters, digits and underscore (_).

Code here: https://github.com/visose/Robots/blob/b711523b2dd5129ecb269acd2589cf4ae3c211ef/src/Robots/FileIO.cs#L326

This restrictions apply to both target attribute names and program names. The reason for program names is because they sometimes get used as variable names inside the robot program.

Some variable names that include the program name can end up as long as: "programName_mechanicalGroupName_000", where mechanicalGroupName is usually "T_ROB1", so that leaves 21 characters for the actual program name.

For old KUKA KRC2 and KRC3 controllers, the file name limit is 24. Robots doesn't know which controller version is being used, so when it exceeds this length and using a KUKA robot, it adds a warning about it.

In Grasshopper, the component's "Create Frame" and "Create Tool" will display a runtime error (shows in a bubble) if the name is invalid. The "Create Program" component will display the error in the Errors output and will not generate the program.

If creating a named target attribute through code, it will throw an ArgumentException if the name passed to the constructor is invalid.

closed 2022-01-18 by visose

robin-gdwl, 2022-01-20:
Amazing! thank you so much! 
I believe that the restriction is perfectly fine as long as the error is clear. It might cause a minor frustration with a few users but will prevent a very big frustration for many others. 
I think the way the errors are handled now is reasonable, and should be straight forward for users to troubleshoot. 

Thanks again for this super quick solution.

visose, 2022-01-21:
I'll publish a new version with this change soon, once I have this other new-user friendly feature in some minimal working state:

[image]

No need to manually download the xml/3dm files from github and copy them to the documents folder anymore (but it's still an option).