visose/Robots #144: [Bug] Robot Libraries window for MacOS

lin-ycv, 2022-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.1.8

### What Rhino version are you using?

Rhino 7 on MacOS

### Current Behavior

Robot Libraries window is not displaying correctly, cannot read names.
[image]

### Expected Behavior

text and icons to be displayed correctly
[image]

### Steps To Reproduce

open Robot Libraries window in MacOS

visose, 2022-03-04:
Thanks. I'll check why the cells of the grid are so narrow. I think the rest of the text not showing up (in the button, title, description) is probably because I have forced a white background and the text color in dark theme is also white. I'll remove any forced styling to avoid this.

visose, 2022-03-06:
[image]

I think this is fixed now. Had to force the height of the cells, since in mac it doesn't compute the correct height (there's an issue open in the eto repo about this).
I had to also force the background color of the panel to be transparent, if not, it uses a gray color rather than the dark theme background.
I fixed also the spinners next to the buttons, they where cropped a bit on mac.

I also noticed the toolbar icons look terrible in mac, and not only the Robots ones. Do people use Rhino/Grasshopper with the dark theme?

visose, 2022-03-06:
@lin-ycv could you check if it looks fixed in the latest release, and if so you can close the issue.

visose, 2022-03-06:
I just noticed another issue with the form in Mac, the icon on the list doesn't change when installing/updating/deleting a library. This also seems like an Eto issue in Mac.

I fixed it by updating the list using a different mechanism, it will go into the next update.

lin-ycv, 2022-03-08:
Sorry for the delayed reply, I don't have ready access to a Mac., but I've just checked and everything seems to be properly formatted now.
The icon issue I don't think is a big deal, will appear as checked after pressing refresh, maybe force a refresh after download?

Most people I know use the dark theme on mac, but I don't think it's that terrible, especially since we usually just use keyword search instead of the toolbar.

Thanks.

closed 2022-03-08 by lin-ycv