visose/Robots #148: [Bug] Joint target creation error (invalid cast String > double[])

ianisl, 2022-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.1.9

### What Rhino version are you using?

Rhino 7 on Windows

### Current Behavior

Hi,

Thanks for the great library.

I am experiencing an issue with the Create Target component in Joint target mode, when supplying the angles via the Joint input. The component yields an `invalid cast String > double[]` error. See attached screenshot and minimum example file.

Important: the error only arises when using a PC with a french version of Windows (which we have to use in our school's lab). On my machine using an EN-US Windows install, everything works fine. Could it be a problem with localization of number formats in Windows (comma is used as a decimal separator in french) ? 

I have tried changing Windows number formatting settings but so far this has not helped with solving the issue.

[image]

[joint-error-example.zip](https://github.com/visose/Robots/files/8222551/joint-error-example.zip)

Thanks!

visose, 2022-03-10:
This a broken backwards compatibility issue. In the latest release the Joints input data type has been replaced from text to a parameter to a custom "Joints" parameter. 

To fix this issue remove the joint input from the target component and add it again, then connect back the the text panel with comma separated values (this will work as before).

I will try to add some check on files saved with an older versions and update this input to the new data type automatically.

ianisl, 2022-03-10:
Thanks for the quick reply. It works with the suggested solution 👍

visose, 2022-03-10:
Reopening the issue until backwards compatibility is fixed, as this might happen often.

visose, 2022-03-11:
The latest version should hopefully be able to open older files without getting this error. If you still get an you can reopen the issue.

closed 2022-03-11 by visose