visose/Robots #165: [Bug] Libraries (GitHub rate limit)

lin-ycv, 2022-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.4.1

### What Rhino version are you using?

Rhino 7 on Windows

### Current Behavior

The plugin seems to be having issues getting the libraries from GitHub, I've tested this on 2 different networks, so it's not my network blocking the connection, after waiting for an hour still the same msg
[image]

### Expected Behavior

Gets the libraries

### Steps To Reproduce

Click on libraries

visose, 2022-06-01:
I can't reproduce it now. Do you remember of anything that have caused it? Like if you had a class with a group of students and they were all refreshing at the same time. I'll try to display a more detailed error of what actually happened.

lin-ycv, 2022-06-01:
Just tested it on my personal computer, and I wasn't able to reproduce it either.
.
it was happening on the lab computers, a few would throw the error while other wouldn't.
I'll have to investigate more next time I'm in the lab.
.
will put this issue on hold for now.

closed 2022-06-01 by lin-ycv