visose/Robots #174: Cant upload via sftp to real robot

robin-gdwl, 2022-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.5.2

### What Rhino version are you using?

Rhino 7 on MacOS

### Current Behavior

I cant seem to send Robot programs to UR-Robots (CB3 Polyscope 3.15) with the FTP option. 
It works with UR-sim with the command: `sftp://ur:easybot@IP/home/ur/ursim-current/programs`.
It does not work on a real robot with `sftp://IP` as well as with any directory or with added username and password. 
The remote connection LOG-output tells me: `ERROR: FTP - the method or operation is not implemented`. 

I can connect to the robots ftp-server via Filezilla without problem. 
What could I be missing? Why is there a difference between Polyscope 3.15 on URSim and a real Robot? 
Where does the error "not implemented" come from? is this part of the plugin or some C# Library. 
Could it be specific to MacOs?
Thanks for any advice

visose, 2022-08-06:
It's likely related to mono, and the library used for sftp distributes a different dll for net framework and netstandard2.

If you don't mind, could you test the following:
 - Close Rhino.
 - Navigate to the Robots plugin folder where all the .dlls files are located, there should be one named `Renci.SshNet.dll`, delete it.
 - Download this package: https://www.nuget.org/api/v2/package/SSH.NET/2020.0.2
 - Open it as if it was a ZIP file, you can do this, for example, by changing the extension to zip.
 - Extract the file `/lib/netstandard2.0/Renci.SshNet.dll` into the Robots plugin folder.
 - Download this second package: https://www.nuget.org/api/v2/package/SshNet.Security.Cryptography/1.3.0
 - Open it as before, and extract the file `/lib/netstandard2.0/SshNet.Security.Cryptography.dll` to the Robots plugin folder.

If you're still getting a not implemented exception, try with the latest Rhino WIP for Mac. They have the switched from mono to NET6.0. I tested the Windows WIP with NET6.0 and it works, but you still need the two netstandard2.0 files mentioned above.

visose, 2022-08-27:
I tested this on Rhino 7 MacOS and URSim, and it works as expected. This is strange because that error seems like a mono issue, and not on any differences between using URSim vs a live robot. The only difference there could be the folder location.
Maybe you have other plugins that contain Renci.SshNet.dll and there is some version conflict.

robin-gdwl, 2022-08-29:
This is indeed the strangest part about this. 
I will check if it makes a difference after uninstalling other plugins.

visose, 2022-09-09:
Hi @robin-gdwl, yesterday I tested the FTP upload with a UR10 and it did work.

There is an issue though, when not specifying the path (sftp://IP) where it will upload correctly to the programs folder but not able to automatically load it due to a wrong path. You can still load it as normal using the pendant. This will be fixed in the next version. Another issue is the URP syntax used does not work with old CB2 models.

But neither of these two issues are related to your error, which seems now more likely that it's specific to your computer.
I'll close the issue for now, but feel free to open if you can reproduce with multiple computers.

robin-gdwl, 2023-04-18:
> If you don't mind, could you test the following:

I finally got another chance to test this. I followed the steps, downloaded the .dll files and replaced the one in the folder and added the other one. 
There was unfortunately no change. The Problem even Persists on a new Macbook (2021). I have not yet used the Rhino WIP. 
Interestingly I tried it on a CB2-UR5 and the ftp-upload actually went through, however the Program can't be opened on the Robot (as you pointed out the URP file is not compatible with older robots as expected).  
This makes the "Not implemented" Issue even weirder and would actually point to there being something odd with the two cb3 robots themself. 
I will test with another laptop tomorrow.

visose, 2023-04-20:
Thanks for testing, looks like either the FTP works or the URP file works for you but never both at the same time 😞

I've attached a version of `Robots.dll` that should display more information about the exception. You can download it here: [Robots.zip](https://github.com/visose/Robots/files/11288985/Robots.zip)

The file to replace is located at: `C:\Users\yourUserName\AppData\Roaming\McNeel\Rhinoceros\packages\7.0\Robots\1.6.5\` (assuming you have updated to the latest version).

If you could send me the full text of the error, hopefully I can find out where it's coming from.

robin-gdwl, 2023-06-01:
I have exchanged the Robots.dll and tried again, here is the error I get from the log: 
```
14:07:37 - Error: FTP - System.NotImplementedException: The method or operation is not implemented.
  at Renci.SshNet.Session.WaitOnHandle (System.Threading.WaitHandle waitHandle, System.TimeSpan timeout) [0x00041] in <b2136f1708424d439013dbe17c5d5f17>:0 
  at Renci.SshNet.Session.WaitOnHandle (System.Threading.WaitHandle waitHandle) [0x0000d] in <b2136f1708424d439013dbe17c5d5f17>:0 
  at Renci.SshNet.Session.Connect () [0x0017b] in <b2136f1708424d439013dbe17c5d5f17>:0 
  at Renci.SshNet.BaseClient.CreateAndConnectSession () [0x00053] in <b2136f1708424d439013dbe17c5d5f17>:0 
  at Renci.SshNet.BaseClient.Connect () [0x0001f] in <b2136f1708424d439013dbe17c5d5f17>:0 
  at Robots.Ftp.Upload (System.Byte[] bytes, System.String fileName, Robots.User user) [0x00049] in <c5f548f2309f4340afd227fcad3d89c6>:0 
  at Robots.RemoteURFtp.UploadFtp (Robots.IProgram program) [0x00022] in <c5f548f2309f4340afd227fcad3d89c6>:0 
  at Robots.RemoteURFtp.Upload (Robots.IProgram program) [0x00000] in <c5f548f2309f4340afd227fcad3d89c6>:0 
```
I have tested with a windows computer and the upload works, so the problem is not on the robot side. 
I am guessing it is some version conflict somewhere. 
Thank you very much for the continued supprt

visose, 2023-06-04:
Thanks @robin-gdwl,
It seems to be due to mono (used on MacOS) not implementing [certain cryptographic algorithms](https://github.com/mono/mono/blob/main/mcs/class/referencesource/System.Core/System/Security/Cryptography/ECDsaCng.cs). 
I've followed a workaround mentioned [here](https://github.com/sshnet/SSH.NET/issues/807). 
Here's an updated `Robots.dll` file that you can test containing the workaround:
[Robots.zip](https://github.com/visose/Robots/files/11646785/Robots.zip)

robin-gdwl, 2023-06-13:
This new .dll file works on my 2021 Macbook!
Thank you very much, this will make working with larger files so much easier.

closed 2023-06-14 by visose