visose/Robots #175: [Bug] Conflict between Rhino3dm and RhinoCommon references

lin-ycv, 2022-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.5.2

### What Rhino version are you using?

Rhino 7 on Windows

### Current Behavior

I'm developing additional custom tool for robots, but the robots nuget package relies on Rhino3dm, which conflicts with RhinoCommon
[image]
and I can't remove RhinoCommon reference since Grasshopper package relies on Rhinocommon.

visose, 2022-08-18:
You might be using the wrong package:
- If you are developing a plugin for Rhino or Grasshopper, use [Robots.Rhino](https://www.nuget.org/packages/Robots.Rhino). This uses RhinoCommon and targets net48.
- If you are developing a standalone or non-Rhino application, use [Robots](https://www.nuget.org/packages/Robots). This uses Rhino3dm and targets netstandard2. There are samples of this package with Unity, Dynamo, WPF.

lin-ycv, 2022-08-18:
Ah, my mistake, did not scroll down on the nuget list, thanks. this makes things a lot easier.

closed 2022-08-18 by lin-ycv