visose/Robots #205: different trajectory on the Staubl robot

visose, 2022-12-15:
### Discussed in https://github.com/visose/Robots/discussions/204

<div type='discussions-op-text'>

<sup>Originally posted by **eduardsik** December 15, 2022</sup>
While testing the Staubli robot, I experience the path deviation shown in the images below, and I cannot find the cause of the error.
The robot executes the program, which in this case is a curve traversal, but about 500 mm higher than what is defined in the simulation. I thought the tcp was set incorrectly, but the amount of sliding upwards is the same in all positions, even if I set the robot plane to negative and place the robot -200mm lower in the simulation.
Another interesting phenomenon is when the robot approaches the track, it always goes down about as much as the difference in height, then goes up and starts to announce the curve. 
Where should I look for the error, what am I not noticing?

The expected curve trajectory:
[image]

The actual trajectory on the real robot:
[image]
</div>

eduardsik, 2022-12-28:
Hello Vince, did you have a chance to uncover the bug?

visose, 2022-12-28:
Hi @eduardsik, sorry holidays got in the way and haven't had a chance yet. As a temporary fix for the time being you could edit the library model to move the robot downwards the distance of the height of the first axis, and adjust the xml. See the Staublis in the Bartlett.3dm/xml files to see what I mean.
As Vince says in the discussion, this should be changed so that you can define a Staubli like the other robots without this issue.

eduardsik, 2023-01-02:
Thanks! I made the changes in xml and 3dm files based on batlett staublis, but somehow the tx60 falls apart. So I will wait for the fix and trying after.

visose, 2023-01-09:
Hi @eduardsik, I just pushed a new update version 1.6.3 that should fix this issue. I haven't tested it yet, if you are able to test  it before I do please let me know if it's working.

visose, 2023-01-09:
> I made the changes in xml and 3dm files based on batlett staublis, but somehow the tx60 falls apart
I forgot to mention to not only move the robot downward by the D value of the first axis but also change the D value in the xml to 0. But hopefully with the new version this is not needed anymore.

eduardsik, 2023-01-10:
Hello Vince,

thank you for the fast bugfix. Now the approaching is good, the robot goes direcly to the starting point of the curve, on the good height. But when I run the code, gives me this error the controller: "VAL3 runtime error. Invalid movement command: invalid destination point or mdesc... (Program name)"
I also run the same program with v1.6.1 and runs without error, but on the wrong height. Only the v1.6.3 gives me this error.
I tried to get closer to the problem, but I did not find any soluion. The program in GH gives me just warnings, but same in both versions (1.6.1 and 1.6.3)

Any idea how to fix it? Thanks, Eduard

visose, 2023-01-11:
Hi @eduardsik, could you test this attached `Robots.dll` file: [Robots.zip](https://github.com/visose/Robots/files/10396312/Robots.zip)
In Windows, close Rhino, go to the folder `C:\Users\{yourUser}\AppData\Roaming\McNeel\Rhinoceros\packages\7.0\Robots\1.6.3\`
and replace the `Robots.dll` file there, then open Rhino again.

eduardsik, 2023-01-12:
Thanks for the very fast bug fixing. After the first test, it seems to work
fine. I'll test and measure out deeply and let you know.

E.

*Eduard Sik*
Creative Engineer <http://about.me/eduardsik> || eduardsik.com ||
***@***.*** || +36 70 237 4151 || book me
<https://doodle.com/eduardsik>

closed 2023-01-18 by visose