visose/Robots #243: [Bug] pulseDO causes Compile Error with UR

KonradJuenger, 2023-08-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.6.5

### What Rhino version are you using?

Rhino 7 on Windows

### Current Behavior

The command pulseDO causes a compile error with the UR5e newest firmware (5.13): "pulseDO000" is not defined. 

### Expected Behavior

a normal pulse

### Steps To Reproduce

```
def Program():
  airbrushTcp = p[-0.02867, -0.03948, 0.13839, 0, 0, 1.59821]
  airbrushWeight = 0.86
  airbrushCog = [0.051, 0.01, -0.058]
  DefaultSpeed = 0.1
  Speed000 = 0.061
  DefaultZone = 0
  Zone000 = 0.002
  Zone001 = 0.002
  PulseDO000 = 0.5
  thread runPulseDO000():
    sleep(PulseDO000)
    set_digital_out(2, False)
  end
  set_tcp(airbrushTcp)
  set_payload(airbrushWeight, airbrushCog)
  movej([0.8541, -1.712, 1.9569, -2.0862, -1.2865, 0.0357], a=3.1416, v=0.3334, r=DefaultZone)
  movej(p[0.16986, -0.62842, 0.27811, 2.47034, 1.06137, -0.17631], a=3.1416, v=0.1347, r=Zone000)
  movej(p[0.15986, -0.62842, 0.27811, 2.47034, 1.06137, -0.17631], a=3.1416, v=0.1253, r=Zone001)
  set_digital_out(2,True)
  run runPulseDO000()
  movej([0.8541, -1.712, 1.9569, -2.0862, -1.2865, 0.0357], a=3.1416, v=0.2209, r=DefaultZone)
end

```

### Anything else?

the error is due to the variable not being defined inside the thread. 
changing this   
```
  PulseDO000 = 0.5
  thread runPulseDO000():
    sleep(PulseDO000)
    set_digital_out(2, False)
  end
```
to 
```
  thread runPulseDO000():
    PulseDO000 = 0.5
    sleep(PulseDO000)
    set_digital_out(2, False)
  end
```
gets rid of the error

visose, 2023-08-18:
Did it work before updating the firmware?

visose, 2023-08-18:
Could you test keeping the variable defined outside of the thread scope but adding the `global` keyword in front:
```
  global PulseDO000 = 0.5
  thread runPulseDO000():
    sleep(PulseDO000)
    set_digital_out(2, False)
  end
  ```

KonradJuenger, 2023-08-18:
Yes this works!

visose, 2023-08-18:
I'll change this on the next version.

visose, 2023-08-22:
The fix should be now available in version 1.6.6

closed 2023-08-22 by visose