visose/Robots #247: Wrong conversion for "get Plane" for UR-Robots

robin-gdwl, 2023-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.6.5

### What Rhino version are you using?

Rhino 7 on Windows

### Current Behavior

Universal robots uses meters as its unit of measurement. 
When converting from a plane to Numbers with the "from plane" component the numbers are accurately converted from millimeters to meters (300 becomes 0.3).
When converting in the other direction with the "get plane" component this conversion does not take place. 
if the input numbers for **x,y,z are 0.3** the resulting plane will be at **(0.3,0.3,0.3)**  not at **(300,300,300)** as it should be.

The plugin should be consistent with the way it converts to and from a manufacturers units. 
Having a conversion in some components but not in others is error-prone in my experience. 

Is this intentional or is it a bug? I can see how it might be the intuitive way for certain workflows. 

[image]

### Anything else?

-

visose, 2023-08-25:
It looks like a bug, I'll fix it in the next release. Thanks for reporting.

visose, 2024-01-08:
Version 1.6.7 is now available with a fix for this.

closed 2024-01-08 by visose