visose/Robots #256: [Bug] FE Panda Python

KonradJuenger, 2023-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.6.6

### What Rhino version are you using?

Rhino 7 on Windows

### Current Behavior

Hi,

I know that FEPanda support is still early -- but here are some bugs I found while testing it with a setup with frankx and an Ubuntu machine:

1. python version
the script is called via `sudo -S python -u` and on my machine this defaults to python 2.7, Frankx expects python 3.7 so I would assume the correct command should be `sudo -S python3 -u`.   Python 2 does not find the frankx library 

2. Incomplete Script definition 
the python script robots generates references the object **dynamic_rel** wich isn't defined. (see below)

3. Motions with more than one target fail 
For some reason any motions with more than one cartesian target produce an error (see below)

Thank you for the continued development of the plugin!

### Anything else?

**2.**
```
from argparse import ArgumentParser
from time import sleep
from frankx import Affine, Robot, JointMotion, PathMotion, WaypointMotion, Waypoint, MotionData, Reaction, Measure, StopMotion, LinearRelativeMotion

def program():
  parser = ArgumentParser()
  parser.add_argument('--host', default='172.16.0.2', help='FCI IP of the robot')
  args = parser.parse_args()
  robot = Robot(args.host)
  robot.set_default_behavior()
  robot.recover_from_errors()
  robot.velocity_rel = 1.0

  DefaultTool = Affine(0, 0, 0, 0, 0, 1, 0)
  DefaultZone = 0

  robot.acceleration_rel = **dynamic_rel**
  robot.jerk_rel = **dynamic_rel**
  data = MotionData(**dynamic_rel**)
  data.velocity_rel = 0.1
  motion = JointMotion([0.10451, -0.00774, 0.15871, -2.44068, 0.00188, 2.43304, 1.83259])

  robot.move(motion, data)

program()

```

**3.**
[image]
[panda_fails_with_multiple_tgt.zip](https://github.com/visose/Robots/files/13752774/panda_fails_with_multiple_tgt.zip)

visose, 2023-12-24:
Hi @KonradJuenger, thanks for testing this. I realize I left out some important info on the summary post, once this is a bit more stable I will make a proper entry into the wiki.

1. Python version
You can change the default python version as shown here:
https://linuxconfig.org/how-to-change-from-default-to-alternative-python-version-on-debian-linux
Using `python3`, even though it narrows it down, it might still open the wrong python version if you have multiple python3 versions installed. The user still must make sure the correct python3 version with the Frankx package is set as default.

2. Currently it is expected that a custom component defining this variable is connected to the "Init commands" input of the CreateProgram comoponent. For example: `dynamic_rel = 0.2`. I'll try to better integrate this, maybe with the CreateSpeed component. I prefer to avoid adding any inputs/components that are specific to a single robot manufacturer.

3. Cartisian motions with the Motion input set to "Joint" are not supported, make sure that they are set to "Linear". I will add a meaninful error when this motion type is set. I think getting an error is better than ignoring the Motion input altogether as a user might get confused on why the robot is moving in a straight line when it's set to Joint. Joint motions using joint values are also supported.

KonradJuenger, 2024-01-04:
Thank you for your prompt response! I've tested the changes after Christmas, and it works. Some further observations:

- FrankX assumes a default TCP at the FE hand gripper, and the conversion to a joint target for the first target doesn't. This results in a jump at the beginning of the movement.
- Speeds appear to be in mm/s for FE, but the robot generates m/s.
- The current installation of FrankX from source as root is quite complex; a wheel file would be very helpful.

This can be closed. If I can assist with any further testing of robots with FE Panda, just drop me a line.

closed 2024-01-04 by KonradJuenger

visose, 2024-01-08:
> FrankX assumes a default TCP at the FE hand gripper
Not sure if the issue is that the robot controller has the TCP set to the hand gripper by default. You can use the desktop interface to change it to an identity matrix (all zeroes except a diagonal of ones).

>something is off the the speeds
Not sure if this is it, but `dynamic_rel` will have an impact on speed, a low value might not allow the robot to accelereate to the set speed.

> you aware of https://github.com/TimSchneider42/franky?
I did not know about it, I'll have a look and see if it's worth using this fork. Thanks!

visose, 2024-01-08:
I just remembered, the speed value that Frankx uses is a percentage of it's max speed (from 0 to 1). Robots speeds are in mm/s. The conversion is done by calculating the time it takes to make the motion using the set mm/s in the simulation divided by the theortically minimum time it could take based on the joint speed limits.