visose/Robots #260: Correct location for local robot XML files [Bug] <title>

nickhfoley, 2024-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.6.X

### What Rhino version are you using?

Rhino 7 for Windows

### Current Behavior

The docs should be updated to reflect that the correct location for custom robots files (at least with Windows 11 / Rhino 7) is actually C:\Users\*username*\AppData\Roaming\McNeel\Rhinoceros\packages\7.0\Robots\libraries

visose, 2024-01-14:
Hi, if you're manually creating or editing 3dm/xml files, you should place them in your Documents/Robots/ folder as it says in the wiki. The folder you mention is just the cache folder used when installing "online" libraries via the UI. You don't need to know about this folder path, it's an implementation detail which could change.

closed 2024-01-14 by visose

nickhfoley, 2024-01-14:
Interesting - I did that first, but nothing happened and grasshopper did
not find the robots. Once I moved the same files to the mentioned folder,
everything worked immediately. Thoughts?

*Nick*

nickhfoley, 2024-01-14:
It seems like it may be a Windows 11 thing, since I just tried the steps on a Windows 10 machine and things worked as expected.

visose, 2024-01-14:
There shouldn't be any difference between Windows 10 and 11.
Double check the path for your `Documents` folder, it might be different.
For example, if you are you are using the OneDrive integration, the path might be: `C:\Users\user\OneDrive\Documents\`

nickhfoley, 2024-01-14:
I think that's the root of the issue - Windows 11 pushes OneDrive so
persistently that even though I'm not "using" OneDrive, and there already
was a Documents folder in the legacy location, it must be looking for the
files in the OneDrive folder.