visose/Robots #264: [Bug] Linear Rail: Data conversion failed from Integer to Joints

lin-ycv, 2024-01-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.6.7

### What Rhino version are you using?

Rhino 8 for Windows

### Current Behavior

number calculated from multiplication isn't accepted, but converting to a string using a panel is ok
[image]

### Expected Behavior

it should be able to take numbers for linear rail

visose, 2024-01-26:
Hi, an implicit cast not being implemented is not really a bug, it would be a feature, but I can add it.
You can already cast the integer to a number (double) instead of a string.

lin-ycv, 2024-01-26:
Thanks, I thought it's a bug because it when you input a number directly from the slider it gets converted, so I thought that logic has been implemented.

visose, 2024-01-26:
That is due to the slider is outputting an number parameter rather than an integer.

visose, 2024-03-23:
Integer to Joint parameter implicit cast should work now. Feel free to reopen if not working as expected.

closed 2024-03-23 by visose