visose/Robots #265: [Bug] Internalized Joint parameters disappear when the param is copied.

robin-gdwl, 2024-01-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.6.7

### What Rhino version are you using?

Rhino 7 for MacOS

### Current Behavior

When copying a "Joints Parameter" with internalized data, the data disappears. 
This also happens when a file with internalized joint data is closed and reopened. 
[image]

### Expected Behavior

The internalised data should be persistent as it is with the other Robots-Parameter.

### Steps To Reproduce

- Create Joint Parameter and fill with values
- internalise the data via right clicking 
- copy the Param with internalised data

visose, 2024-01-31:
I never got to finish implementing persistent parameters, so I wouldn't rely on internalizing any Robots parameters. I should've turned them to normal parameters a long time ago to disallow internalizing. Data will be lost on internalized params after saving, closing and opening the document again.

The problem is that the Robots data types are not designed to be (de)serialized. This is particularly important for objects that reference to other objects.

I think at least I should get the Joints parameter to serialize as this is a very simple data type.

The problem with other ones that seem simple like Speed, Zone, Frame, is that Robots relies on reference equality to check which attributes are shared among different targets. If the same attribute gets serialized in multiple places, the instances will stop having reference equality after deserialized (they are now two different objects in memory).

I'm actually starting a project soon that uses Robots as a library and will likely benefit from (de)serializing these, so I might have a go at implementing it properly.

robin-gdwl, 2024-01-31:
Thank you for the explanation. 
Makes sense, and the workaround should be easy enough for the joint param. I can simply internalise it as a text parameter.

visose, 2024-03-23:
Joint parameters should be able to be internalized in version 1.7.0. Feel free to reopen if not working.

closed 2024-03-23 by visose