visose/Robots #284: Robot tab does not show up in Rhino 8 after installing this plugin

arasbm, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What Robots version are you using?

1.8.0

### What Rhino version are you using?

Rhino 8 for MacOS

### Current Behavior

using package manager plugin is found and installed without any error. When looking in "installed" tab I can see the plugin is supposedly installed. However there is no *Robots* tab after restarting Rhino and Grasshopper. Tried uninstall and install and restarting several times with no luck. I am not sure how to get any more information as there are no errors, but I would really like to try this plugin. Any help is appreciated.

I am using a Mac with M3 chip (ARM64) 

### Expected Behavior

Robot tab should be shown in grasshopper after install

### Steps To Reproduce

No new tab is added and no error is presented. In the package manager the Robots plugin appears to be in the install section

### Anything else?

Thank you!

visose, 2024-07-03:
Could you share a screenshot of the Installed tab of the package manager?

arasbm, 2024-07-03:
Yes of course, here is the screenshot:

[image]

The notification about changes does not go away no matter how many times I restart. Also, you can see in the libraries folder only` Pufferfish....gha` is present. Should there not be a file in there for the Robots plugin?

arasbm, 2024-07-03:
I am really sorry to have bothered you, it seems that I just needed to restart the computer. I restarted Rhino several times but I guess that was not enough to load the new plugin. But after I shutdown and restarted the Mac it seems to have been loaded:
[image]

Thank you for your help and your work on this plugin. Looking forward to using it!

closed 2024-07-04 by arasbm