visose/Robots #325: [Bug] "Define Joints" different behaviour than expected

robin-gdwl, 2025-05-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### What Robots version are you using?

1.8.0

### What Rhino version are you using?

Rhino 8 for MacOS

### Current Behavior

The `Define Joints` component has some unexpected behaviour. 
I would have expected it to create a list of comma-seperated joint values in radians according to the degrees set in the inputs. 
However, it appears that there are some conversions being done (sometimes 90° sometimes 180°) 
Is this specific to a certain manufacturer? 
If I get current Joint values from a universal Robot and set them in the define joints component, the resulting pose is different than the actual robot. 

[image]

### Expected Behavior

`Define Joints`should just convert degrees to radians and create a joint param.

visose, 2025-05-06:
Different plugin 😄 

[image]

closed 2025-05-06 by visose

robin-gdwl, 2025-05-06:
Oops, sorry about that. 
Did not check and thought it was part of Robots. 
Thanks for the reply.